bvavra/MIDI_Accordion #2: Problem with schematics: interal pull-up resistor in Arduino is too weak

mojca, 2017-08-12:
First of all, let me express a deep gratitude for providing so accurate description of the project. I blindly ordered a bunch of opto interruptors (ITR-9608) for [our project](https://wiki.mittelab.org/progetti/mojca/accordion), but I could not get them working by following your schematic and it took me quite a bit of exploring until we realized where the problem was.

This schematic finally worked:
[image]

It turned out that Arduino's internal 50k (or whatever the exact value is) internal pull up is simply too weak. The problem was solved for me after adding additional 1k resistors next to Arduino's diginal inputs and configure the ports as regular input rather than `INPUT_PULLUP`. We then even managed to use Arduino nano for the accordion thanks to your idea rather than having to depend on Mega (even though we'll probably still use Mega at the end).

I accidentally bought two hundred 470 Ohm resistors instead of cca. 180-270 Ohm (it was the shop's fault) and ended up with cca. 1 V when the switch was OFF, but that still translated to nice zeroes at the digital input.

I wonder how that schematic could have worked for you unless your opto interruptors were slightly different.

bvavra, 2017-08-12:
Hiyo,

Wow, your project looks amazing, and I'm wildly impressed you were able to do so much in a single week (including building the box itself)!  And thank you for adding my project to your credits. :)

Re: your issue - I'm surprised that the Arduino's internal pullup resistors (which I believe are 20k Ω) didn't work for you, because I also used the ITR-9608 and didn't have a problem.  The only guess I have is based on your schematic above that says the signal is going to a digital pin - although my code is doing a digital read against the port, the interruptor is actually connected to an analog pin, not digital, so that difference may possibly be the source of your issue.  I had tried connecting the interruptor to a digital pin during development and it did not work properly, but I did not try experimenting with the pullup resistor value or anything else to fix it, and instead went forward with using the analog pins.

mojca, 2017-08-12:
No, the box itself was lasercut upfront because bringing the laser to the camp was not realistically doable.

I actually tested with analogue inputs and kept reading the analogue value. The difference was there, but it was around 20 units out of 1024 if I recall correctly. Note that I was using both ITRs and Arduino nano clones ($5) from eBay. Maybe exact characteristics are slightly different from originals. After switching to 1k resistors I started getting really nice jumps from 110 to 960 (0.5 to 5 V) and it worked on both analogue and digital ports. (After connecting everything we realised that nano doesn't support digitalRead on A6 and 7.)

I can repeat the experiment on Mega's analogue pin though.

bvavra, 2017-08-13:
Hmm - I can't imagine the difference should be so significant between the Mega and a Nano clone, but I can't say for sure.

I tested one of my ITRs against [my initial prototype](https://github.com/bvavra/MIDI_Accordion/tree/master/Prototypes/Opto_Interruptor/Opto_Interruptor_1) on an Arduino Uno and had no problem getting good reads with this setup:

[image]

[image]

```
//ITR blocked
D9 HIGH: Value for analog PIN: 1012
Value for analog PORT: 11000000

D9 LOW: Value for analog PIN: 1011
Value for analog PORT: 11000000

//ITR unblocked
D9 HIGH: Value for analog PIN: 35
Value for analog PORT: 11000100

D9 LOW: Value for analog PIN: 162 //a bit of a jump here, but not enough to flip the bit value
Value for analog PORT: 11000000 
```

I also tested this with a digital pin instead of analog and the digital port reads did still work correctly, to my surprise, so perhaps that's not the issue you're having.  

Sidenote: the transistor is not required for a single interruptor (you could write the pin HIGH and LOW from the LED anode instead), but I included it in this sketch because it is required to drive down the current when reading 8 of them at a time, and I wanted to get it set up correctly so I had less moving parts when changing how many rows and/or columns of interruptors I had hooked up.

mojca, 2017-08-14:
Please give me some time to do some further testing and measurements (and to repeat the theory of transistors to try to come up with potential explanation). It could also be that transistors used in non-original ITR9608 switches could be different just enough to cause this effect. (I had the same effect when I replaced my own/external 1k pull-up with a 100k pull-up resistor.)

A local electronics store would need 3-5 weeks to deliver these optointerruptors, eBay/aliexpress with DHL was much faster and I have no clue where they come from.

mojca, 2017-08-15:
I went through some more consistent testing and I suspect I know what went wrong.

In the initial testing with super limited time that I had available the code for reading registers with Nano didn't work out of the box on nano (or maybe the wiring was wrong or whatever else), so I wrote all the code myself. I must have used
```
pinMode(3, INPUT_PULLUP);
```
instead of
```
pinMode(A3, INPUT_PULLUP);
```
To read from A3.

That made the values jump between roughly 1014-1023 with pretty high noise, so distinguishing the values would be nearly impossible.

When I added the external pull-up resistor it started working because the pin was configured for normal input by default.

In today's repeated test using a 220 Ohm resistor for the diode and 100k resistor as pull-up on the transistor side leads to jumps from roughly 15 to 1010 which is perfect. It also works just fine with the internal pull-up once I configured it properly.

Once we had the "PCB" ready and the correct code in place, I didn't try to change the schematic and my colleague had enough time with other aspects of the project, so he never even checked what I did (well, he did spend one full day with wire-wrapping tool :).

Sorry for the wrong diagnosis. And many many many thanks for the wonderful project description which made our project & experience a lot less painful.

closed 2017-08-15 by mojca

bvavra, 2017-08-16:
No problem - I'm happy you figured it out and got everything working.  Sometimes you just need a [rubber duck](https://en.wikipedia.org/wiki/Rubber_duck_debugging) to talk to. :)

I'm glad my documentation was helpful to you.  It was my goal to not only build a working MIDI accordion, but to also document my work well enough to make this project as accessible as possible to others, and you are proof of that success.  If there's anything you think I can do to better document or explain certain aspects of the project, please let me know.