bvavra/MIDI_Accordion #5: bmp180

denia85, 2019-12-07:
Good afternoon.You wrote that it is possible to implement disabling bmp180 in the future .Could you point me how to do this

bvavra, 2019-12-09:
Hiyo,

See https://github.com/bvavra/MIDI_Accordion/blob/master/MIDI_Accordion/MIDI_Accordion.ino line 43 - BMP is disabled by default; you must uncomment this line to enable it.  This requires recompiling the code.  If you wanted to add a switch to turn it on and off at runtime, some code changes will be required (such as replacing the #define with a real variable and hooking to up to be toggled on and off).

bvavra, 2019-12-11:
From our email conversation:

> Yes about a more in-depth guide on how to do this. Could you share your experience?

Okay, I understand now.  That's correct, I haven't created a full page on this yet in my [in-depth guide](http://bvavra.github.io/MIDI_Accordion/overview/) yet; I will leave this issue open to do this.  

In the meantime, I have basic steps in the [readme's quick guide](https://github.com/bvavra/MIDI_Accordion#guide), and you can follow the [Sparkfun tutorial](https://learn.sparkfun.com/tutorials/bmp180-barometric-pressure-sensor-hookup-) to get started on how to program for the BMP.  You can also check out [the prototypes I made](https://github.com/bvavra/MIDI_Accordion/tree/master/Prototypes/BMP) to start integrating using the BMP180 for MIDI control.

bvavra, 2019-12-24:
Ok, I added a [detailed page on using the BMP180](http://bvavra.github.io/MIDI_Accordion/volume/) - please let me know if this makes sense or if you have any other questions.

closed 2019-12-24 by bvavra