bvavra/MIDI_Accordion #7: Accordion midi - change opto interruptor

TheMaurice, 2020-04-15:
Hi, my name is Maurizio. I write from Rome. I saw your project and I really like it. I congratulate you. I have a Paolo Soprani 120 that works very well. I want to leave it like this but also add your midi circuit. Can I replace the Opto Interruptor (ITR-9608) with Hall Effect Reed Magnetic Sensors without changing the code? Thank you very much. (translated with Google)

bvavra, 2020-04-15:
Hello,

Thank you!  Trying to add the MIDI capability while leaving the reeds in place is quite the challenge. 
 I personally have not looked into using hall sensors, so I cannot say with 100% certainty that replacing the optos with them will work without code updates.  At the very least, I will say I am performing digital reads on the optos, so the hall sensors you use will need to be able to send digital output signals (or at least, the Arduino will need to interpret the signal via digital read).  Otherwise you will need to update the code to use the analog pins instead of digital pins.  Either way, so long as you have the matrix of sensors set up the same way, no other significant changes should need to be made - only pin mapping.

Good luck with your project!

TheMaurice, 2020-04-17:
thank you so much for replying. the sensors that I plan to use are these https://it.wikipedia.org/wiki/Reed_(dispositivo). they are simple normally open on-off hall effect switches. now here in Italy we have a lot of time to stay at home. I had seen your project for a long time and wanted to carry it out. maybe this is the right time. Greetings. Maurizio

closed 2020-05-02 by bvavra