bvavra/MIDI_Accordion #8: Great Project - But - Bluetooth/Hairless/Lag is over 100ms

bwinterstine, 2020-04-28:
I re-used your schematics and code to build a midi accordion out of an old Excelsior accordion.  I really appreciate all the work you put into this! 

My setup is: Midi Accordion, Genuine Arduino Mega, Bluetooth HC-05, 38400 Baud Rate ---> 2015 Macbook Pro with Hairless Serial App to Mainstage (for sounds).  Inside Mainstage, I set it up like this:  midi channel 5 for left hand Bass Notes, midi channel 4 for left hand Chords and midi channel 3 for right hand piano keys.  This all works well.

Problem:  When I play a note on the the accordion, any note, it takes approximately 120ms for the sound to be played.  I have attached a screenshot from Audacity.  I had recorded the sound of me hitting a note on the accordion and and the note being played on my Mac.  I used Audacity to analyze the time delay.  Also, I use Mainstage with USB Midi Keyboards and do not have any problems with latency on this Macbook.

Does anyone else have this problem?

[image]
[image]

bvavra, 2020-04-28:
Hmm.  My knowledge on the Mac side is a bit lacking, so I can't comment on if anything is slowing it down on that end.  I'm reading that Mac Core Audio is not known to experience added latency (unlike Windows, where I had to use ASIO4All to make my accordion playable), so I'm going to assume there's no issue there.  Some other possible things to look into:

1. Have you tried using a USB connection instead of Bluetooth to see if the latency still exists over USB?
2. You mention using a baud rate of 38400 - have you tried using 115200?
3. Are you using the pressure sensor code at all?  If so, can you try turning it off to see if that has any noticeable impact (in theory it should only add 2-4ms).

bwinterstine, 2020-04-28:
Wow, fast reply!  Thanks.

I can use either the USB connection or bluetooth.  The USB connection has a latency of 86ms which is slightly better; I did try it.  I have tried other baud rates but the latency stays about the same.

I am not using the pressure sensor code, yet.  I want to get the basic midi accordion working well first.  My next step is to add some knobs and buttons to send midi control information to Mainstage.  The last step will be the pressure sensor.

I have the code stripped down to a bare minimum and I'm sure it is not contributing the the latency (your code works great and is pretty darn fast).

I think the problem is in hairless midi software compiled on the latest 64 bit MacOS.  I have reported a problem there also.

bvavra, 2020-04-28:
Awesome - sounds like you're doing pretty much the same thing I was.

If you determine Hairless MIDI to be the culprit, you could try cutting it out completely by updating the Arduino to send data directly as MIDI using [mocoLUFA](https://github.com/kuwatay/mocolufa).  Not sure how well that will work over bluetooth, but it should work fine over USB - that way the accordion would be recognized by the OS as a MIDI device.  I haven't tried this personally, but others who have done this and similar projects went this route with great results.

bwinterstine, 2020-04-29:
I found out today that my midi keyboard (Berhinger U-Control) has latency times of around 63ms.  This is way too high and I must have a problem in my MAC.  The same keyboard connected directly to my Korg Kronos workstation (via midi) has almost no detectable/measurable latency; around 10ms.  I need to do some investigation into my MAC and what it is doing to add all this latency.

closed 2020-04-29 by bwinterstine

bwinterstine, 2020-04-30:
I have determined that a large amount of the latency is in the MacBook
Pro.  The interface does go into sniff mode after a few seconds but the
latency continues when in active mode.   When in sniff mode, the first midi
data can take over 200ms.

I am thinking about buying a used Roland sound module but they would not
support the wireless bluetooth midi interface.

What were you using to receive bluetooth midi from the HC-05?

Bruce

bvavra, 2020-05-01:
> What were you using to receive bluetooth midi from the HC-05?

I had just bought a standard bluetooth USB dongle and configured it using the Windows device configuration:

[image]

See also [Step 4 in this tutorial](https://www.instructables.com/id/Cheap-2-Way-Bluetooth-Connection-Between-Arduino-a/) for greater details.

> I am thinking about buying a used Roland sound module but they would not
support the **wireless bluetooth midi** interface.

I know Windows does not support this natively (which is why I don't use mocoLUFA), but I think OSX does. Something that might be worth trying (if you're not using this already) is the [Yamaha MD-BT01](https://usa.yamaha.com/products/music_production/accessories/md-bt01/index.html), which adds wireless bluetooth midi capabilities for OSX/iOS devices.

I recommend reaching out to @JasonBugeja to see if he ran into any latency issues like this - he has [his own MIDI Accordion projects](https://github.com/JasonBugeja?tab=repositories) that connect to OSX/iOS over bluetooth using the Yamaha MD-BT01, so he may be able to better help you with Mac compatibility than I can.

bwinterstine, 2020-05-01:
OK.  I think I’m going to build a Master bluetooth midi thru device to take the accordion bluetooth midi and convert it to a standard midi Tx DIN connector.  Then I can simply plug this into any midi compatible device and play the accordion thru it.  I will test it with my Korg Kronos and if it works well I will pick up an older Roland sound module.  Then, I can tailor some knobs and buttons on the accordion to change patches, volume and effects. 

From all my testing I have discovered that my testing method has an inherent 20-30ms latency from the time the key is touched to the time the switch/contact senses the key press.  This is true of most instruments and this has to be subtracted from my previous test results.  Even an acoustic piano has a 20-30ms latency that really is never discussed.

After approximately 30 separate tests and measurements, I arrived at the following conclusions and estimates:

1.  The measurement technique I used has an inherent 30ms latency the numbers below show the actual latency measurements after this was subtracted.
2.  All direct USB connections to the MacBook are playable and the additional latency (from accordion software and MacBook processing) is between 11 to 50ms.
3.  Additional latency above 50ms becomes very noticeable.  This is the value to strive for (to keep latency less that this).
4.  MainStage adds a great deal of latency as compared to a freeware sound engine, SFZ.  See the light blue tests below.  MainStage does not appear to run well on my 2015 MacBook Pro 13” with 8 GB Ram 2.7 i5 processor.  MainStage appears to add approximately 30-40ms latency when playing through the MacBook Pro core audio (no external audio interface).  I have included a snapshot of what MainStage claims to be the resultant latency of the settings (48kHz, 128 buffer size, safety buffer on).  Their screen says the latency is 27.9ms but I am not sure how accurate that number is.
5.  Blutooth seems to add around 20ms of latency over and above the USB/Hairless combination.   But, there are untested variables in this estimate.  For example, is Hairless-Serial working differently and causing more or less latency?  Is the MacBook Bluetooth interface causing more or less latency?  Is the HC-05?  Tough to measure more accurately.
6.  Bluetooth midi to my MacBook Pro through Hairless-Serial is not playable due to 74ms of additional latency (from all factors).