bvavra/MIDI_Accordion #9: Resistor count wrong in doc?

RosesTheN00b, 2020-05-14:
Hey :)
what a cool project. Thank you so much for sharing!
I am currently trying out whether this can be a great raspberry pi project.

In your [overview page](http://bvavra.github.io/MIDI_Accordion/overview/) there are 11x 1k listed. But on the schematic I can only find 9 (8 next to the transistor, 1 bluetooth). Did I forget any somewhere?

Greetings

Sebastian

RosesTheN00b, 2020-05-14:
Created PR #10 for easy fix :-)

bvavra, 2020-05-14:
Thanks - great catch!  I ran through my docs and yeah, can't find where I got that number from.  My only guess is I was playing around with more voltage dividing somewhere and got my numbers mixed up.  Pull request has been merged.

PR #10 (2020-05-14): change number of resistors in overview
Fixes #9
closed 2020-05-14 by bvavra