bvavra/MIDI_Accordion #11: 6 channels

denia85, 2020-12-15:
Hi, how to add in unison to channel 1 channel 4. To channel 2 channel 5. To channel 3 channel 6. as a result, you should get 6 channels. I'm not asking you to do it for me just show me the direction

bvavra, 2020-12-15:
Hi denia,

The code can easily be modified to do this - specifically, adding two lines of code near the end of the note_midi function at the bottom of [MIDI_Accordion.ino](https://github.com/bvavra/MIDI_Accordion/blob/master/MIDI_Accordion/MIDI_Accordion.ino) to send multiple noteOn and noteOff commands for multiple channels.  Let me know if this puts you in the right direction or if you'd like another hint. ;)

denia85, 2020-12-15:
> Привет Дения,
> 
> Код можно легко изменить для этого - в частности, добавив две строки кода ближе к концу функции note_midi в нижней части [MIDI_Accordion.ino](https://github.com/bvavra/MIDI_Accordion/blob/master/MIDI_Accordion/MIDI_Accordion.ino) для отправки нескольких команд noteOn и noteOff для нескольких каналов. Дайте мне знать, если это приведет вас в нужное русло или если вы хотите получить еще один намек. ;)

Thanks for the answer. but my knowledge is not enough. Tell me more about which lines and where. for example, 1 and 4 channels. I'll do the rest myself

denia85, 2020-12-15:
> Hi denia,
>
> The code can easily be modified to do this - specifically, adding two lines of code near the end of the note_midi function at the bottom of [...]

https://github.com/bvavra/MIDI_Accordion/issues/11#issuecomment-745542524

denia85, 2020-12-15:
Thanks for the answer. but my knowledge is not enough. Tell me more about which lines and where. for example, 1 and 4 channels. I'll do the rest myself

bvavra, 2020-12-15:
Change this:

```
if(on) {
  MIDI.sendNoteOn(pitch, midi_vel, channel);
}
else {
  MIDI.sendNoteOff(pitch, midi_vel, channel);
}
```

to this:

```
if(on) {
  MIDI.sendNoteOn(pitch, midi_vel, channel);
  MIDI.sendNoteOn(pitch, midi_vel, channel+3);
}
else {
  MIDI.sendNoteOff(pitch, midi_vel, channel);
  MIDI.sendNoteOff(pitch, midi_vel, channel+3);
}
```

denia85, 2020-12-15:
And the channels themselves 4 5 6 where?

bvavra, 2020-12-15:
This is done automatically, because the channel variable is set to 1,2, or 3.

denia85, 2020-12-15:
Please

denia85, 2020-12-15:
That is enough? Thanks

closed 2020-12-15 by denia85