bvavra/MIDI_Accordion #12: Several keyboard layout options

denia85, 2020-12-18:
Hi. tell me how to make several keyboard layout options. with a different tone. Five down five up

byte table = 5;
const char left_notes_midi_numbers[10][7][8] = {
  {
    {0,0,0,0,0,0,0,0},//53
    {0,0,0,0,0,0,0,0},//52
    {0,0,0,0,0,0,0,0},//
    {0,0,0,0,0,0,0,0},//
    {0,0,0,0,0,0,0,0},//
    {0,0,0,0,0,0,0,0},
    {0,0,0,0,0,0,0,0}//
  },
  {
    {0,0,0,0,0,0,0,0},//53
    {0,0,0,0,0,0,0,0},//52
    {0,0,0,0,0,0,0,0},//
    {0,0,0,0,0,0,0,0},//
    {0,0,0,0,0,0,0,0},//
    {0,0,0,0,0,0,0,0},
    {0,0,0,0,0,0,0,0}//
  },
  {
    {0,0,0,0,0,0,0,0},//53
    {0,0,0,0,0,0,0,0},//52
    {0,0,0,0,0,0,0,0},//
    {0,0,0,0,0,0,0,0},//
    {0,0,0,0,0,0,0,0},//
    {0,0,0,0,0,0,0,0},
    {0,0,0,0,0,0,0,0}//
  },
  {
    {0,0,0,0,0,0,0,0},//53
    {0,0,0,0,0,0,0,0},//52
    {0,0,0,0,0,0,0,0},//
    {0,0,0,0,0,0,0,0},//
    {0,0,0,0,0,0,0,0},//
    {0,0,0,0,0,0,0,0},
    {0,0,0,0,0,0,0,0}//
  },
  {
    {44,37,39,42,46,47,41,40},//53
    {46,45,36,38,58,43,48,36},//52
    {57,56,56,57,55,58,56,59},//
    {82,48,85,49,54,50,53,40},//
    {52,72,73,51,76,50,79,49},//
    {45,48,101,63,59,47,58,69},
    {52,72,73,51,55,58,56,59}//
  },
  {
    {0,0,0,0,0,0,0,0},//53
    {0,0,0,0,0,0,0,0},//52
    {0,0,0,0,0,0,0,0},//
    {0,0,0,0,0,0,0,0},//
    {0,0,0,0,0,0,0,0},//
    {0,0,0,0,0,0,0,0},
    {0,0,0,0,0,0,0,0}//
  },
  {
    {0,0,0,0,0,0,0,0},//53
    {0,0,0,0,0,0,0,0},//52
    {0,0,0,0,0,0,0,0},//
    {0,0,0,0,0,0,0,0},//
    {0,0,0,0,0,0,0,0},//
    {0,0,0,0,0,0,0,0},
    {0,0,0,0,0,0,0,0}//
  },
  {
    {0,0,0,0,0,0,0,0},//53
    {0,0,0,0,0,0,0,0},//52
    {0,0,0,0,0,0,0,0},//
    {0,0,0,0,0,0,0,0},//
    {0,0,0,0,0,0,0,0},//
    {0,0,0,0,0,0,0,0},
    {0,0,0,0,0,0,0,0}//
  },
  {
    {0,0,0,0,0,0,0,0},//53
    {0,0,0,0,0,0,0,0},//52
    {0,0,0,0,0,0,0,0},//
    {0,0,0,0,0,0,0,0},//
    {0,0,0,0,0,0,0,0},//
    {0,0,0,0,0,0,0,0},
    {0,0,0,0,0,0,0,0}//
  },
  {
    {0,0,0,0,0,0,0,0},//53
    {0,0,0,0,0,0,0,0},//52
    {0,0,0,0,0,0,0,0},//
    {0,0,0,0,0,0,0,0},//
    {0,0,0,0,0,0,0,0},//
    {0,0,0,0,0,0,0,0},
    {0,0,0,0,0,0,0,0}//
  }
};

void setup()
{
  button_minus.setClickHandler(click);
  button_plus.setClickHandler(click);
  
void loop()
{
  button_minus.loop();
  button_plus.loop();

#ifdef DEBUG
      Serial.print("Note ");
      if(on){
        Serial.print("on: ");
      }
      else {
        Serial.print("off: ");
      }
      Serial.println(pitch);
    #else
 if(on) {
  MIDI.sendNoteOn(pitch, midi_vel, channel);
  MIDI.sendNoteOn(pitch, midi_vel, channel+3);
}
else {
  MIDI.sendNoteOff(pitch, midi_vel, channel);
  MIDI.sendNoteOff(pitch, midi_vel, channel+3);
}       
    #endif
  }

}

void click(Button2& btn) {
    if (btn == button_minus) {
      if(table > 0){
        table--;
        #ifdef DEBUG
          Serial.println(table);
        #endif
      }
    } else if (btn == button_plus) {
      if(table < 9){
        table++;
        #ifdef DEBUG
          Serial.println(table);
        #endif
      }
    }
}

denia85, 2020-12-18:
it should be like this .How to make a toggle the left and right variables with a single button
const char left_notes_midi_numbers[][8] = {
  {38,39,57,C5,0,63,0,51},//A0
  {61,Eb5,64,F5,0,0,0,0},//A1 0.0.64.0.0.0
  {66,67,68,69,48,53,0,0},//A2 0.67.0.0.69
  {Bb5,0,85,61,88,0,97,40},//A3
  {0,72,73,38,F3,0,Eb3,37},//A4
  {45,53,0,63,Bb2,32,0,30},//A5
  {99,0,43,63,0,C3,0,0},//A6
  {76,0,79,0,0,0,67,0}//A7
};

const char left_notes_midi_numbers_0[][8] = {
  {38,39,57,71,57,63,0,51},//A0
  {60,62,63,64,0,55,0,0},//A1 0.0.66.0.0.0
  {65,66,67,68,65,53,0,59},//A2 0.63.0.0.68
  {69,0,85,49,88,50,97,40},//A3
  {0,72,73,38,52,0,50,48},//A4
  {45,53,53,63,57,55,53,53},//A5
  {99,69,43,63,0,59,0,65},//A6
  {76,0,79,0,0,0,67,0}//A7
};

const char left_notes_midi_numbers_1[][8] = {
  {0,0,0,0,0,0,0,0},//
  {0,0,0,0,0,0,0,0},//
  {0,0,0,0,0,0,0,0},//
  {0,0,0,0,0,0,0,0},//
  {0,0,0,0,0,0,0,0},//
  {0,0,0,0,0,0,0,0},//
  {0,0,0,0,0,0,0,0},//
  {0,0,0,0,0,0,0,0}//
};

bvavra, 2020-12-18:
Hello,

To do this, you need a new variable that keeps track of which layout you are using:

`int right_key_layout = 0;`

Then, in the loop() function, you need code that scans your button for input and, if the given button is pressed, change the value of the selected layout.  Here is some pseudo code (you may need to adjust for debounce as necessary - there are many Arduino tutorials on how to read button inputs you can use for better examples):

```
int yourButtonPin = 6;
void loop() {
  
  uint8_t val = digitalRead(yourButtonPin);
  if (val == LOW)//button is pressed
  {
    right_key_layout++;
    if(right_key_layout > maxNumberOfLayouts)
      right_key_layout = 0;
  }
  
  //keyboard scanning code...
}
```

Then finally, you can use the variable when selecting which notes to play in the note_midi() function:
`pitch = right_notes_midi_numbers[right_key_layout][group][position];`

You can do the same thing for both left and right hands.

denia85, 2020-12-18:
and could you write what and where exactly to write. I've done a lot of work with your code. that's all that's left.

denia85, 2020-12-18:
please

denia85, 2020-12-18:
It should be like this?
int right_key_layout = 0;
char right_hand_pins[] = { 3, 4, 5, 6, 7, 8, 9, 10, 11 };
// array to store up/down status of right keys
int RightKeysStatus[] = {  
  B0000000,
  B0000000,
  B0000000,
  B0000000,
  B0000000,
  B0000000,
  B0000000,
  B0000000,
  B0000000
};
//Note: Based on how the opto-interruptors are laid out,
//You will likely have to remap their pitch numbers (53-93)
const char right_notes_midi_numbers[][8] = {
  {82,85,88,82,85,87,82,84},//4 крайний сверху ряд баяна с низу вверх 8 кнопок с права на лево
  {79,81,64,67,57,51,49,54},//11 средний ряд баяна с низу вверх 8 кнопок с права на лево
  {53,58,56,61,60,65,63,68},//5 средний ряд баяна с низу вверх 8 кнопок с права на лево
  {66,72,70,75,73,78,77,82},//6 верхний ряд баяна с низу вверх первые 2 кнопоки с права на лево
  {80,85,84,42,47,54,44,57},//7 нижний ряд баяна с низу вверх первые 8 кнопок
  {65,62,62,59,62,55,56,51},//8 нижний ряд баяна с низу вверх вторые 8 кнопок
  {77,73,74,70,71,67,68,64},//9 нижний ряд баяна с низу вверх 2 кнопоки
  {68,67,86,64,83,61,80,58}//10 средний ряд баяна с низу вверх 8 кнопок с права на лево
  
};

bvavra, 2020-12-18:
[image]

bvavra, 2020-12-18:
See above where the changes should be made.  To clarify, the button scanning code may not be correct - please use whatever code makes sense for the button you chose, and look up Arduino tutorials on how to scan it accordingly.

denia85, 2020-12-18:
Thanks I'll try

bvavra, 2021-01-06:
I've not heard any updates in a bit, so I assume this has been addressed.

closed 2021-01-06 by bvavra